adrianomarto/soft_uart #1: Where is raspberry_gpio.c ?

irfanreza, 2017-09-21:
Hello,
Related to that title, i want to ask where is **raspberry_gpio.c** file ? I can't find that.
Can you help me ? thx.

adrianomarto, 2017-09-23:
Hi,

This is my mistake - I am sorry for that. The file raspberry_gpio.c is no longer necessary.
I had initially created the raspberry_gpio.c to implement an interface with the GPIO. But afterwards, I found some functions defined in linux/gpio.h that do exactely what I was trying to do, but in a more complete and standard way. So I eliminated the file raspberry_gpio.c and started using the linux/gpio.h.

However, if you still want to take a look at raspberry_gpio.c then you can do it by inspecting the change history. There are not many changes, so I believe you will not have problems to find is.

Cheers,
Adriano.

closed 2017-09-25 by adrianomarto

bbinet, 2018-06-05:
The README says: `Edit the file raspberry_gpio.c to set the correct value for BCM2708_BASE, according to the version of the Raspberry Pi you are using.`
```
// For Raspberry Pi v2 and v3:
#define BCM2708_BASE  0x3f000000

//For Raspberry Pi v1:
#define BCM2708_BASE  0x20000000
```
So there is no need anymore to #define BCM2708_BASE ?

adrianomarto, 2018-06-10:
That is correct. It was necessary for a previous implementation. Now, you need to build it on the same kind platform you are going to use. It means that, if you are going to use it on a RPi3, then you need to build it on a RPi3.

The kernel headers have all the correct parameters needed to access the GPIO.