adrianomarto/soft_uart #3: Invalid module format

ESLHemantSangle, 2018-03-25:
There is a error while using "insmod soft_uart.ko" command on Raspberry PI 3

ERROR
insmod: ERROR: could not insert module soft_uart.ko: Invalid module format

OS Detials
PRETTY_NAME="Raspbian GNU/Linux 8 (jessie)"
NAME="Raspbian GNU/Linux"
VERSION_ID="8"
VERSION="8 (jessie)"
ID=raspbian
ID_LIKE=debian

so can you help me solve this error
Thanks,

adrianomarto, 2018-03-26:
That usually happens when we build the module using kernel headers of a different version from the kernel being used. Have you built the module in the same system into which you are trying to load it? Have you updated the kernel?

ESLHemantSangle, 2018-03-26:
Thanks for the reply @adrianomarto ,
Yes actually I got that and also upgraded the Linux kernel headers but the make file was still trying to find Linux-headers-4.4.50-v7+ so I downloaded them from web and installed it but its still giving me the same error.

And Yes I am using the same Raspberry PI Model 3 to built and load the module.(using the setups given by you on the https://github.com/adrianomarto/soft_uart )

ESLHemantSangle, 2018-03-30:
Hello @adrianomarto can you please help me out for the solution. As I need this in my production.
Thanks,

adrianomarto, 2018-03-31:
Hi @ESLHemantSangle,
I would try the following:
1. Type "uname -a" on the command prompt and take note of the indicated kernel version.
2. Go the folder where you are building the module and type "make clean" and then "make". The printouts will indicate the kernel headers' version used to build the module. That must be exactly the same number as indicated in step 1.
If you find that the versions are different then you found the problem.

You can also try to load the module using "insmod ./soft_uart.ko" to make sure you are loading the module in the current folder instad of something in the search path.
You can also use the command modinfo to verify the module parameters.
If that does not help, please send the output of all those commands. That can help me to understand what is going on.

Cheers,
Adriano.

ESLHemantSangle, 2018-03-31:
Hi @adrianomarto 
I did as you suggested but still the error is same please find the output of steps given by you below

1. `uname -a` command output
`root@raspberrypi:/home/pi/soft_uart# uname -a`
`Linux raspberrypi 4.4.50-v7+ #970 SMP Mon Feb 20 19:18:29 GMT 2017 armv7l GNU/Linux`

2. `make clean` command output
`root@raspberrypi:/home/pi/soft_uart# make clean`
`make -C /usr/src/linux-headers-4.4.50-v7+ M=/home/pi/soft_uart clean`
`make[1]: Entering directory '/usr/src/linux-headers-4.4.50-v7+'`
  `CLEAN   /home/pi/soft_uart/.tmp_versions`
  `CLEAN   /home/pi/soft_uart/Module.symvers`
`make[1]: Leaving directory '/usr/src/linux-headers-4.4.50-v7+'`

3. `make` command output
`root@raspberrypi:/home/pi/soft_uart# make`
`make -C /usr/src/linux-headers-4.4.50-v7+ M=/home/pi/soft_uart modules`
`make[1]: Entering directory '/usr/src/linux-headers-4.4.50-v7+'`
`  CC [M]  /home/pi/soft_uart/module.o`
`  CC [M]  /home/pi/soft_uart/raspberry_soft_uart.o`
`  CC [M]  /home/pi/soft_uart/queue.o`
`  LD [M]  /home/pi/soft_uart/soft_uart.o`
`  Building modules, stage 2.`
`  MODPOST 1 modules`
`  CC      /home/pi/soft_uart/soft_uart.mod.o`
`  LD [M]  /home/pi/soft_uart/soft_uart.ko`
`make[1]: Leaving directory '/usr/src/linux-headers-4.4.50-v7+'`

4. `make install` command output
`root@raspberrypi:/home/pi/soft_uart# make install`
`sudo install -m 644 -c soft_uart.ko /lib/modules/4.4.50-v7+`
`sudo depmod`

5. `insmod soft_uart.ko` command output
`root@raspberrypi:/home/pi/soft_uart# insmod soft_uart.ko `
`insmod: ERROR: could not insert module soft_uart.ko: Invalid module format`

6. `insmod ./soft_uart.ko` command output
`root@raspberrypi:/home/pi/soft_uart# insmod ./soft_uart.ko`
`insmod: ERROR: could not insert module ./soft_uart.ko: Invalid module format`

7. `modinfo soft_uart.ko` command output
`root@raspberrypi:/home/pi/soft_uart# modinfo soft_uart.ko`
`filename:       /home/pi/soft_uart/soft_uart.ko`
`version:        0.1`
`description:    Software-UART for Raspberry Pi`
`author:         Adriano Marto Reis`
`license:        GPL`
`srcversion:     583A908CEF3BB3E6F1F7BE3`
`depends:`        
`vermagic:       4.4.50-v7 SMP mod_unload modversions ARMv7 `
`parm:           gpio_tx:int`
`parm:           gpio_rx:int`

As the output of step 1,2 and 3 referring to same kernel version i.e. `linux-headers-4.4.50-v7+` but still the module is not starting.

Thanks,
Hemant

adrianomarto, 2018-03-31:
Hi @ESLHemantSangle,

Have you noticed that the version indicated by modinfo is 4.4.50-v7 whilst your kernel is 4.4.50-v7**+**? I don't know exactly from where you downloaded the kernel headers, but it is not exactly the same version as the kernel you are using.

I would try to use the linux kernel and linux-headers from raspbian repository instead of downloading them from anywhere else.

Just some tips:
1. The plus at the end of the version indicates that the source files have been modified.
2. Do not use rpi-update. It created a mess in my system in the last time I tried to use it. Use apt-get instead.

Cheers,
Adriano.

ESLHemantSangle, 2018-03-31:
Thank very much @adrianomarto with your help I have solved the issue of kernel and the soft_uart module is running now thank you so much for your code and time

adrianomarto, 2018-03-31:
I am glad that the problem is solved.

ESLHemantSangle, 2018-04-04:
Hi @adrianomarto,
I am trying to connect GSM module as PPP modem to ttySOFT0 but the connection is failing. I tried steps given by you to test the SOFT UART and they are working but still the connection is failing. So is there any way I can test the SOFT UART using different steps if yes then please let me know.

Thanks,
Hemant

Tangerino, 2018-09-25:
How can we load the driver for 2 devices such as SOFT0 and SOFT1 (and select the appropriated pins) ? As both will run on very low baud rate I believe we will not face any performance issue. Thanks

ollnixon, 2020-08-06:
I am having the same issue as @ESLHemantSangle .

The initial problem was a mismatch of headers as described in #17 , but following a firmware upgrade and package update, this was resolved. I'm now receiving an "Invalid module format" when running the `insmod` command.

1. `uname -a` output
  ```
Linux piz2 5.4.51+ #1332 Tue Aug 4 18:28:38 BST 2020 armv6l GNU/Linux
```
2. `make clean` output
  ```
make -C /usr/src/linux-headers-5.4.51+ M=/home/pi/soft_uart clean
make[1]: Entering directory '/usr/src/linux-headers-5.4.51+'
  CLEAN   /home/pi/soft_uart/Module.symvers
make[1]: Leaving directory '/usr/src/linux-headers-5.4.51+'
```
3. `make` output
  ```
make -C /usr/src/linux-headers-5.4.51+ M=/home/pi/soft_uart modules
make[1]: Entering directory '/usr/src/linux-headers-5.4.51+'
  CC [M]  /home/pi/soft_uart/module.o
  CC [M]  /home/pi/soft_uart/raspberry_soft_uart.o
  CC [M]  /home/pi/soft_uart/queue.o
  LD [M]  /home/pi/soft_uart/soft_uart.o
  Building modules, stage 2.
  MODPOST 1 modules
  CC [M]  /home/pi/soft_uart/soft_uart.mod.o
  LD [M]  /home/pi/soft_uart/soft_uart.ko
make[1]: Leaving directory '/usr/src/linux-headers-5.4.51+'
```
4. `make install` output
  ```
sudo install -m 644 -c soft_uart.ko /lib/modules/5.4.51+
sudo depmod
```
5. `sudo insmod soft_uart.ko` output
  ```
insmod: ERROR: could not insert module soft_uart.ko: Invalid module format
```
6. `modinfo soft_uart.ko` output
  ```
filename:       /home/pi/soft_uart/soft_uart.ko
version:        0.2
description:    Software-UART for Raspberry Pi
author:         Adriano Marto Reis
license:        GPL
srcversion:     FF97B9BF705FF2E8754580E
depends:
name:           soft_uart
vermagic:       5.4.51+ mod_unload modversions ARMv6 p2v8
parm:           gpio_tx:int
parm:           gpio_rx:int```

GopikaSPillai, 2024-09-26:
I am having the same issue of invalid module format while trying to insert a module
can anyone help me

GopikaSPillai, 2024-09-26:
Sir,
can you help me i am having the same issue.

> uname -a
Linux gen-u22 6.8.0-45-generic #45~22.04.1-Ubuntu SMP PREEMPT_DYNAMIC Wed Sep 11 15:25:05 UTC 2 x86_64 x86_64 x86_64 GNU/Linux

> modinfo hello_world.ko
filename:       /home/gen/linux/hello_world.ko
version:        2:1.0
description:    A simple hello world driver
author:         EmbeTronicX <embetronicx@gmail.com>
license:        GPL
srcversion:     221E551EB4C73B5888265D9
depends:        
retpoline:      Y
name:           hello_world
vermagic:       6.8.0-45-generic SMP preempt mod_unload modversions