adrianomarto/soft_uart #7: Multiple instances

ijustwant, 2018-10-11:
Hi
I need more than one soft uart on my Raspberry pi. How can i do that?
Tried to rename file and tty before compilation with no luck.

adrianomarto, 2018-10-16:
Hi,
The module was not designed to do that. Renaming the file would not make any difference either.
If you really want to have multiple serial ports, then you need to do some code modifications. It is not trivial, but it is possible.

MatrixMonk, 2019-03-21:
hi, i need multiple as well, what should i change in the code to make it work? 
I looked in the code ... i have seen few places like module.c ... it that be enough?
-erik

ijustwant, 2019-04-09:
I do not have the knowledge on how to do this. Does anyone take the challenge? :)

mertozal, 2020-08-11:
I need multiple software uart too for use sim800l. Is there any guide or way to make that happen? Can you help us a little please @adrianomarto

wolfgangmauer, 2020-10-09:
I could use that too.

Axel21g, 2020-11-01:
Hi, 
#### I found not a solution, but a workaround for two instances of soft_uart on two different pairs of pins at my raspberry Pi 3B.
Thanks you so much for your work adrianomarto, which inspired me (see offtopic). As he mentioned in 2018 this code is not designed to several instances.

So i changed the code at five positions to get a new driver that can be additionally installed. This is no good kernel-module programming - you get a new major number for the new device. However it is my first trial and error work with modules and it works fine for me. 

Just add some "x" at the correct places:
#### Standard git clone
`git clone https://github.com/adrianomarto/soft_uart`
#### Make a copy of the original code in the directory soft_uartx (look at the x at the end)
`cp -r soft_uart/ soft_uartx/`
#### compile standard driver 
```
cd soft_uart
make
sudo make install
```
#### normal call of insmod with the new pair of non-standard GPIOs
`sudo insmod soft_uart.ko gpio_tx=20 gpio_rx=21`
#### check the device and the (major/minor) number of the device
```
ls /dev/ttySOFT0
file /dev/ttySOFT0
```
### new driver (with "x" at the end)
```
cd
cd soft_uartx
```
#### EDIT module.c
```
line 114-OLD:   soft_uart_driver->driver_name           = "soft_uart";
line 114-NEW:   soft_uart_driver->driver_name           = "soft_uartx";

line 115-OLD:   soft_uart_driver->name                  = "ttySOFT";
line 115-NEW:   soft_uart_driver->name                  = "ttySOFTx";
```
#### EDIT Makefile
```
line 1-OLD:     obj-m += soft_uart.o
line 1-NEW:     obj-m += soft_uartx.o
line 3-old:     soft_uart-objs := module.o raspberry_soft_uart.o queue.o
line 3-new:     soft_uartx-objs := module.o raspberry_soft_uart.o queue.o

line 15-old:    sudo install -m 644 -c soft_uart.ko /lib/modules/$(RELEASE)
line 15-new:    sudo install -m 644 -c soft_uartx.ko /lib/modules/$(RELEASE)
```
#### compile new driver in new directory soft_uartx
```
make
sudo make install
```
#### insmod of the modified soft_uartx.ko
`sudo insmod soft_uartx.ko`
#### check the news device `/dev/ttySOFTx0` and the (major/minor) number of the device.
the major number should be different from the one above. This is why this is a workaround, not a solution.
```
ls /dev/ttySOFTx0
file /dev/ttySOFTx0
```
### Proof with hardware
Connect GPIO 20 (TXD-1) to GPIO 27 (RXD-2) and GPIO 21 (RXD-1)  to GPIO 17 (TXD-2). Therefore the two serials are connected with a "crossover" cable.

`minicom -D /dev/ttySOFTx0`
and via ssh from another computer
`minicom -D /dev/ttySOFT0`

type some info in the minicom-terminals, the result should be received in the other terminal resp.

workes :-) Hopefully there is no mistake in the above code! Remember, it is my first approach to kernel module programming.
 
### After a reboot do not forget:
```
sudo depmod
sudo insmod ~/soft_uart/soft_uart.ko gpio_tx=20 gpio_rx=21
sudo insmod ~/soft_uartx/soft_uartx.ko
```
### Offtopic: Why did i need this? Show how serial works with baud-rate of 1!
I need two softserials with extreme low baud-rates. (1 baud) to make serial communication visible to students. Additionally to the above proof with hardware, two LED with resistors at each TXD-line were connected. Now the students can see the serial-communication at very low speed. One can see with their eyes whether the LSB oder MSB is sent first by choosing an appropriate character e.g. "@".
### How can one get such a low baud-rate like 1 Baud?
In file raspberry_soft_uart.c : 

Line 126/127 change " baudrate" to "1". This is baud-rate of 1 is fixed now and cannot be changed any more.
```
period = ktime_set(0, 1000000000/1);
gpio_set_debounce(gpio_rx, 1000/1/2);
```
#### Thank you once more @adrianomarto

user21a, 2020-11-04:
Hi, 
#### I found not a solution, but a workaround for two instances of soft_uart on two different pairs of pins at my raspberry Pi 3B.
Thanks you so much for your work adrianomarto, which inspired me (see offtopic). As he mentioned in 2018 this code is not designed to several instances.

So i changed the code at five positions to get a new driver that can be additionally installed. This is no good kernel-module programming - you get a new major number for the new device. However it is my first trial and error work with modules and it works fine for me. 

Just add some "x" at the correct places:
#### Standard git clone
`git clone https://github.com/adrianomarto/soft_uart`
#### Make a copy of the original code in the directory soft_uartx (look at the x at the end)
`cp -r soft_uart/ soft_uartx/`
#### compile standard driver 
```
cd soft_uart
make
sudo make install
```
#### normal call of insmod with the new pair of non-standard GPIOs
`sudo insmod soft_uart.ko gpio_tx=20 gpio_rx=21`
#### check the device and the (major/minor) number of the device
```
ls /dev/ttySOFT0
file /dev/ttySOFT0
```
### new driver (with "x" at the end)
```
cd
cd soft_uartx
```
#### EDIT module.c
```
line 114-OLD:   soft_uart_driver->driver_name           = "soft_uart";
line 114-NEW:   soft_uart_driver->driver_name           = "soft_uartx";

line 115-OLD:   soft_uart_driver->name                  = "ttySOFT";
line 115-NEW:   soft_uart_driver->name                  = "ttySOFTx";
```
#### EDIT Makefile
```
line 1-OLD:     obj-m += soft_uart.o
line 1-NEW:     obj-m += soft_uartx.o
line 3-old:     soft_uart-objs := module.o raspberry_soft_uart.o queue.o
line 3-new:     soft_uartx-objs := module.o raspberry_soft_uart.o queue.o

line 15-old:    sudo install -m 644 -c soft_uart.ko /lib/modules/$(RELEASE)
line 15-new:    sudo install -m 644 -c soft_uartx.ko /lib/modules/$(RELEASE)
```
#### compile new driver in new directory soft_uartx
```
make
sudo make install
```
#### insmod of the modified soft_uartx.ko
`sudo insmod soft_uartx.ko`
#### check the news device `/dev/ttySOFTx0` and the (major/minor) number of the device.
the major number should be different from the one above. This is why this is a workaround, not a solution.
```
ls /dev/ttySOFTx0
file /dev/ttySOFTx0
```
### Proof with hardware
Connect GPIO 20 (TXD-1) to GPIO 27 (RXD-2) and GPIO 21 (RXD-1)  to GPIO 17 (TXD-2). Therefore the two serials are connected with a "crossover" cable.

`minicom -b 1200 -D /dev/ttySOFTx0`
and via ssh from another computer
`minicom -b 1200 -D /dev/ttySOFT0`

type some info in the minicom-terminals, the result should be received in the other terminal resp.

workes :-) Hopefully there is no mistake in the above code! Remember, it is my first approach to kernel module programming.
 
### After a reboot do not forget:
```
sudo depmod
sudo insmod ~/soft_uart/soft_uart.ko gpio_tx=20 gpio_rx=21
sudo insmod ~/soft_uartx/soft_uartx.ko
```
### Restrictions
There are no problems sending continuous data from soft_uart to PC an vice versa e.g. with minicom as long as the size of the continous data is below the TX-buffersize (256 byte in queue.h). I have problems sending continous data from the soft_uart to my pc when the data length is greater than 256 bytes (tx_buffersize). There is no problem sending the same data from PC to RASPI.
This is an issue  with the original software at my RASP 3B+ with brand new raspbian. ( [see issue #20](https://github.com/adrianomarto/soft_uart/issues/20)) 
Of course, the same problem occours with each of the two soft_uarts as well.

TODO:
The kernel-messages have not been adapted, they should be `printk(KERN_INFO "soft_uartx: TEXT);` at every occurrence in all `*.c` files.

### Offtopic: Why did i need this? Show how serial works with baud-rate of 1!
I need two softserials with extreme low baud-rates. (1 baud) to make serial communication visible to students. Additionally to the above proof with hardware, two LED with resistors at each TXD-line were connected. Now the students can see the serial-communication at very low speed. One can see with their eyes whether the LSB oder MSB is sent first by choosing an appropriate character e.g. "@".
### How can one get such a low baud-rate like 1 Baud?
In file raspberry_soft_uart.c :  line 126/127 
```
period = ktime_set(0, 1000000000/baudrate);
gpio_set_debounce(gpio_rx, 1000/baudrate/2);
```
change " baudrate" to "1". This is baud-rate of 1 is fixed now and cannot be changed any more by software. It is for demonstrating purpose only!
```
period = ktime_set(0, 1000000000/1);
gpio_set_debounce(gpio_rx, 1000/1/2);
```
#### Thank you once more @adrianomarto