adrianomarto/soft_uart #14: Cannot allocate memory

biswaKL, 2019-09-19:
Hi, While running below command
`sudo insmod soft_uart.ko`

I  am getting error like:
`insmod: ERROR: could not insert module soft_uart.ko: Cannot allocate memory`

**This is in Debian buster latest raspbian.**

ahmad7091, 2020-04-18:
I have the same problem. No solution?

dtyuterev, 2020-07-26:
I have the same problem. No solution?

rohitsd1409, 2023-10-06:
check dmesg.
stack trace will be printed.

yccgabriel, 2025-05-20:
I bumped into the same issue, on a Raspberry Pi Zero W Rev 1.1,
(after fixing some obvious compilation error)

the problem was I need to add a 512 offset to the GPIO PIN numbers.
so for example: `static int gpio_tx = 17 + 512;` and `static int gpio_rx = 27 + 512;`

this offset apparently varies across raspberry pis.
hope that helps.  :)