adrianomarto/soft_uart #29: Compilation error on RPI 4 (incompatible-pointer-types)

d1ckb31n, 2024-03-26:
Any idea on how to fix this compilation error on a RPI4?

pi@raspberrypi:~/soft_uart $ make
make -C /usr/src/linux-headers-6.6.20+rpt-rpi-v8 M=/home/pi/soft_uart modules
make[1]: Entering directory '/usr/src/linux-headers-6.6.20+rpt-rpi-v8'
  CC [M]  /home/pi/soft_uart/module.o
/home/pi/soft_uart/module.c:47:22: error: initialization of ‘ssize_t (*)(struct tty_struct *, const u8 *, size_t)’ {aka ‘long int (*)(struct tty_struct *, const unsigned char *, long unsigned int)’} from incompatible pointer type ‘int (*)(struct tty_struct *, const unsigned char *, int)’ [-Werror=incompatible-pointer-types]
   47 |   .write           = soft_uart_write,
      |                      ^~~~~~~~~~~~~~~
/home/pi/soft_uart/module.c:47:22: note: (near initialization for ‘soft_uart_operations.write’)
/home/pi/soft_uart/module.c:52:22: error: initialization of ‘void (*)(struct tty_struct *, const struct ktermios *)’ from incompatible pointer type ‘void (*)(struct tty_struct *, struct ktermios *)’ [-Werror=incompatible-pointer-types]
   52 |   .set_termios     = soft_uart_set_termios,
      |                      ^~~~~~~~~~~~~~~~~~~~~
/home/pi/soft_uart/module.c:52:22: note: (near initialization for ‘soft_uart_operations.set_termios’)
cc1: some warnings being treated as errors
make[3]: *** [/usr/src/linux-headers-6.6.20+rpt-common-rpi/scripts/Makefile.build:248: /home/pi/soft_uart/module.o] Error 1
make[2]: *** [/usr/src/linux-headers-6.6.20+rpt-common-rpi/Makefile:1938: /home/pi/soft_uart] Error 2
make[1]: *** [/usr/src/linux-headers-6.6.20+rpt-common-rpi/Makefile:246: __sub-make] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-6.6.20+rpt-rpi-v8'
make: *** [Makefile:9: all] Error 2
pi@raspberrypi:~/soft_uart $

bibiboss, 2024-05-01:
Same here on RPI5

Prafulcoder, 2024-10-30:
open module.c file ... replace static void soft_uart_set_termios(struct tty_struct*, struct ktermios*); with static void soft_uart_set_termios(struct tty_struct*,const struct ktermios*);  in both function definition and derivation ..