adrianomarto/soft_uart #31: Request for Missing Headers for Kernel Version 6.1.21-v8+

Prafulcoder, 2024-10-30:
Hello,

I am currently working on a project with a Raspberry Pi and need the kernel headers for version 6.1.21-v8+. Unfortunately, I am encountering issues as the headers are not available, and I need them to compile a module.

**Details:**
- Raspberry Pi Model: RPI4
- Current Kernel Version: 6.1.21-v8+
- Errors Encountered:
- make -C /usr/src/linux-headers-6.1.21-v8+ M=/home/vishnu/soft_uart modules
make[1]: *** /usr/src/linux-headers-6.1.21-v8+: No such file or directory.  Stop.
make: *** [Makefile:9: all] Error 2
  
Could you please provide the missing headers for this version? It would greatly help in resolving the issues I am facing.

Thank you for your assistance!

Best regards,  
Prafulcoder

th-grimaldi, 2025-05-06:
bump, I have a similar similar issue.

* hardware: Raspberry Pi Zero 2W
* distro version: Ubuntu 24.04.2 LTS Server
* kernel version: 6.8.0-1024-raspi
* error:

```bash
~/soft_uart$ make -C /usr/src/linux-headers-6.8.0-1024-raspi M=/home/mike/code/soft_uart modules
make[1]: *** /usr/src/linux-headers-6.8.0-1024-raspi: No such file or directory.  Stop.
make: *** [Makefile:9: all] Error 2
```

th-grimaldi, 2025-05-06:
oh the fix for this is really simple: `sudo apt install linux-headers-6.8.0-1024-raspi`.

you can probably close this issue.